yaqwsx/KiKit #295: DRC exclusions ignored

sslupsky, 2022-01-24:
**Describe the bug**: A clear and concise description of what the bug is.
Encountered a situation when using the fabrication option to generate manufacturing data.  There was a court yard violation that I made an exclusion for in the GUI.  When I ran kikit, kikit ran a DRC and did not comply with the exclusion and failed to generate output because of the error. 
```
** Found 1 DRC violations: **
[courtyards_overlap]: Courtyards overlap Severity: error
  Local override
  @(194.1000 mm, 116.6000 mm): Footprint J25
  @(208.1000 mm, 117.2000 mm): Footprint H3
DRC failed. See report above. No files produced
```

**Expected behavior**: A clear and concise description of what you expected to happen.
DRC should observe exclusions and not terminate data generation.

yaqwsx, 2022-01-24:
This will be a little tricky - KiCAD doesn't give us an API to see the DRC exclusions. So we will probably have to parse the file and locate the excluded rules.

yaqwsx, 2022-05-05:
This was implemented in 5ef4768070b43beabf6ce319ad6581880bd02148 and ca5a408057f86d4349d8b8e8cd6eb1a3381b425d.

closed 2022-05-05 by yaqwsx

yaqwsx, 2022-05-06:
Also, KiKit panelization now correctly handles DRC exclusions and transplants them to the panel (implemented in 7503cf6af0add7e03deb02966d706f6496a6a51f)