yaqwsx/KiKit #296: KiKit launch prevents kikad.exe from closing

WhiteChairFromIkea, 2022-01-24:
# Description
kicad.exe process does not terminate after using KiKit.

# To Reproduce
See video

1) Start kicad, File -> Open demo project -> choose "USB stickhub..";
2) Launch PCB, close PCB, close KiCad;
3) observe no `kicad` process is left;
4) Launch kicad, launch PCB editor, Toolse -> Extr. plugins -> KiKit;
5) Close Kikit, close pcbnew (why does it ask for file save??), close kicad;
6) kicad.exe process keeps running in the background.

Bug at step 6 cannot be seen while using other python plugins.

# Screenshots

https://user-images.githubusercontent.com/46539653/150784367-a3d23bc8-544b-4f84-94ef-f06478d9a396.mp4

# Version

Specify which version of KiKit do you use:
```
 - kikit, version 1.0.2;
 - PCM installation on 2022-Jan-24 (+cli installation finalization with pip - install... as described by kikit first time prompt);
 - Windows 10 x64 always up to date.
```

yaqwsx, 2022-02-12:
The problem was in improperly destroyed dialog, this should be fixed now in [bb47edc](https://github.com/yaqwsx/KiKit/commit/bb47edc5f46a1acc4988bb07eb23457c9560d5ac).

closed 2022-02-12 by yaqwsx

sslupsky, 2022-02-24:
I do not believe this problem is entirely resolved.  I experienced another KiCad crash today and I have the master branch with the commit mentioned earlier installed.