yaqwsx/KiKit #299: Text using KiCad Variables not panelized

esden, 2022-01-29:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

kikit, version 1.0.2

### KiCAD version

Version: (6.0.1), release build

### Operating system

Arch Linux

### Description

Silkscreen Text items that use KiCad variables are not being duplicated in the panezilation.

I would recommend to expand those text variables in the panelization step and insert the expanded text variables in the panel *.kicad_pcb.

### Steps to Reproduce

Create a Silkscreen text item with the content `${REVISION}`. This will insert the Revision field from the "Page Settings"->"Title Block"->"Revision" field on the silkscreen inside of KiCad.

After panelization all those Silkscreen Text items disappear.

yaqwsx, 2022-01-29:
This seems like a duplicate of #289. Please, try the upstream version.

yaqwsx, 2022-02-03:
I assume this is resolved; if not please feel free to reopen.

closed 2022-02-03 by yaqwsx