yaqwsx/KiKit #300: Example Failes

greenscreenflicker, 2022-01-31:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

kikit, version 1.0.2

### KiCAD version

6.0.1

### Operating system

Windows 11

### Description

I try to reproduce the example the following example
https://github.com/yaqwsx/KiKit/blob/master/doc/examples.md
with the attached file.
[original.zip](https://github.com/yaqwsx/KiKit/files/7970771/original.zip)

I get the following error.

`C:\Users\mchhe\OneDrive\Dokumente\git\rs232iso_v3\pcb_rc5_2\panel>kikit panelize  --layout 'grid; rows: 2; cols: 2; space: 2mm'  --tabs 'fixed; width: 5mm' --cuts 'mousebites; drill: 0.5mm; spacing: 1mm; offset: 0.2mm' original.kicad_pcb panel.kicad_pcb
Usage: kikit panelize [OPTIONS] INPUT OUTPUT
Try 'kikit panelize --help' for help.

Error: Got unexpected extra arguments (cols: 2; space: 2mm' width: 5mm' drill: 0.5mm; spacing: 1mm; offset: 0.2mm' original.kicad_pcb panel.kicad_pcb)`

### Steps to Reproduce

Open file attached, run command from example, observe error.

yaqwsx, 2022-01-31:
The example (even with your file) works:

[image]

I suspect that you have a problem with your terminal - it seems that it interprets a semicolon in the command as the end of the command. I am not entirely familiar with windows; try escaping the semicolon or use double quotes instead of single quotes.

greenscreenflicker, 2022-01-31:
Double Quotes worked for me. I propose to change this in the examples to fix this.

yaqwsx, 2022-01-31:
Changing it in examples is not possible - this is shell-dependent and especially on UNIX systems using single quotes is beneficial as it allows you to enter, e.g., texts. I will add notes about this into the documentation.

closed 2022-01-31 by yaqwsx