yaqwsx/KiKit #301: Warning

greenscreenflicker, 2022-01-31:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

kikit, version 1.0.2

### KiCAD version

6.0.1

### Operating system

Windows 11 (German)

### Description

I get the following error.

[image]

### Steps to Reproduce

Run the example from

greenscreenflicker, 2022-01-31:
Run this to reproduce >kikit panelize  --layout "grid; rows: 2; cols: 2; space: 2mm"  --tabs "fixed; width: 2mm" --cuts "mousebites; drill: 0.5mm; spacing: 1mm; offset: 0.2mm" original.kicad_pcb panel.kicad_pcb

Cancel works for me.

yaqwsx, 2022-01-31:
This is a bug in KiCAD - there have popped a few issues regarding this in KiCAD issue tracker. Hopefully, this gets fixed by 6.0.2 release.

yaqwsx, 2022-03-25:
Since the fix is not present in KiCAD's upstream, I tried to mitigate the problem in c282fa88f70114b3a63c47f5ed0136b8473718bf. Please, let me know if it helps.

yaqwsx, 2022-03-28:
Just tested on Windows 10, seems to work just fine.

closed 2022-03-28 by yaqwsx