yaqwsx/KiKit #302: Missing components

s56jsc, 2022-01-31:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

1.0.2

### KiCAD version

6.0

### Operating system

windows 10

### Description

After runing kikit some components are missing. Do I do something wrong or it is bug?
[image]
[image]
[image]
[image]
[image]

best regards
Simon

### Steps to Reproduce

As you can look at picture there missing on LED and ESP32 components

yaqwsx, 2022-01-31:
Did you really read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)? Especially the part about [missing components](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md#kikit-throws-away-components-from-my-panel-how-can-i-prevent-it)?

s56jsc, 2022-02-01:
My mistake. I ididn't read well FAQ. Solution is to use tolerance command