yaqwsx/KiKit #303: Prolong not working when using a negative offset

MiguelAReis, 2022-02-03:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

1.0.2

### KiCAD version

6.0.1

### Operating system

Kubuntu 21.04

### Description

When using a negative offset on the mousebites, positive values of prolong don't seem do do anything. Example image below:
[image]

### Steps to Reproduce

kikit panelize \
    --layout 'grid; rows: 9; cols: 6;  space :2mm' \
    --tabs 'fixed; vcount:0;' \
    --cuts 'mousebites; drill: 0.2mm; spacing: 0.5mm; offset: -0.2mm; prolong: 5mm' \
    --framing 'railslr; width: 4mm; space: 2mm; cuts: v' \
    --post 'millradius: 1mm' \
    LED4Bit.kicad_pcb panel.kicad_pcb

yaqwsx, 2022-02-03:
KiKit adds only the holes that penetrate the substrate. This is why the extra holes do not show up.

MiguelAReis, 2022-02-03:
There still seems to be some space in the substrate that is able to be penetrated but kikit doesn't seem to recognize there is.

yaqwsx, 2022-02-03:
The holes you marked should get just slightly behind the outline, therefore they do not get rendered. It seems reasonable to bloat the substrate by e.g., hole radius (https://github.com/yaqwsx/KiKit/blob/5e5494c6e64dd47fb4b47605048bf8fc0d82c444/kikit/panelize.py#L938). Therefore any hole that touches the board outline will get rendered.

MiguelAReis, 2022-02-09:
Ok thanks

closed 2022-02-09 by MiguelAReis