yaqwsx/KiKit #304: Big text dropped by panelize

eiffel31, 2022-02-04:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

kikit, version 1.0.2

### KiCAD version

6.0.1

### Operating system

Linux debian stable

### Description

Big texts on silkscreen layers are ignored. If the size is reduced then they are kept.
[bug_text.zip](https://github.com/yaqwsx/KiKit/files/8006516/bug_text.zip)

### Steps to Reproduce

Using the provided file:
kikit panelize -p :jlcTooling    --layout 'grid; rows: 2; cols: 2; space: 1mm'     --tabs 'fixed; width: 5mm'     --cuts 'mousebites; drill: 0.5mm; spacing: 1mm; offset: 0.3mm' test.kicad_pcb test_panel.kicad_pcb

pcbnew test_panel.kicad_pcb
The big "44" are gone.

yaqwsx, 2022-02-04:
The text has some margin and for the "44" it reaches out of the board outline. This is why it is not included. Introduce a tolerance: `--source 'tolerance: 10 mm' and it works. This is covered in the FAQ "KiKit throws away my components", however, since the text bounding box is not visible in KICAD I get this can be confusing.

closed 2022-02-04 by yaqwsx

eiffel31, 2022-02-06:
That's a very clear explanation, thank you for that.

As you noticed, it is an unexpected behavior because the user is not aware of the bounding box of the text.

So, do you think that text could have a specific rule? I can make 2 proposals specific to text handling:
- Reduce text bounding box of 1/3 of the box size before checking if the text fit the board outline.
- If more than half of the text bounding box surface is inside the board, then keep the text.

yaqwsx, 2022-02-06:
This shouldn't be text-specific and it shouldn't contain any magic number. Currently, KiKit works like KiCAD when you create a selection of the board with tolerance offset from and you drag it from top left to bottom right. Only components that are fully in it will be included. What might make sense is to create a second mode, where KiKit would behave just like when you draw the selection box from the bottom right to top left.