yaqwsx/KiKit #305: nightly docker fails

lazyzero, 2022-02-05:
Hello,

the latest and nightly docker images fails for me with the message: 

>  ../src/common/stdpbase.cpp(62): assert "traits" failed in Get(): create wxApp before calling this 

I tried with an project that was working for the old image, with the same cli call in the container.

With an old image, that I can start by image id, it is all running smooth. I am still on Kicad 5.1.12 on Kubuntu 20.04.3 LTS

Any idea what could be wrong?

Best Christian

yaqwsx, 2022-02-05:
This seems to be a problem in KiCAD 6.0.1. The bug has been reported. It is shown, however, it should not affect the result.

PS: Your KiCAD version does not matter; docker images are always built against the current stable KiCAD (which is 6.0.1 at the moment). That also means that you won't be able to open the final panels in your KiCAD. Consider building a custom docker image if you would like to use KiCAD 5

lazyzero, 2022-02-05:
Hello Jan,

thanks for the reply. I made a fork and tried to build it against the ppa for the 5.1 releases of Kicad. 
Works like a charm. Thanks for the hint.

Christian

closed 2022-02-05 by lazyzero