yaqwsx/KiKit #306: Panelize or fab JLCJLCJLCJLC aware

eiffel31, 2022-02-06:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

JLCPCB has one option: the user can specify the location of the JLCPCB identifier. This is done while adding a text "JLCJLCJLCJLC" with specific size on the board.

The problem: if such text is put on some board, then panelize tool usage will result in many instances of this specific text.

Two proposal:
- panelize could keep this text for one board only in the panel
- fab tool, when used with jlcpcb, could remove all extra instances of this text to just keep a single one

lazyzero, 2022-02-09:
All designs I panelized and send to JLCPCB had a frame and JLCPCB used the frame for their identifier, fidusials and tooling holes.
If you want to be sure you can put the "JLCJLCJLCJLC"-text on the frame

Best Christian

yaqwsx, 2022-03-08:
How should the user specify, which text to keep? Besides the UI for selecting the text, the feature is pretty simple to implement; however, I see limited use for it - I think the rail solution is much cleaner.

eiffel31, 2022-03-09:
Keeping only one is enough. As the initial model is replicated many times, keeping the text on any of them seems to give the same result.

I do agree that designs using rail would better use it for this order identifier. For my first order, I removed the text and put it on the rail.

The case is when you design a PCB with a specific place for this identifier. Then if you are happy with the result and want to panelize the PCB, then you will have this text many times.

troisiemetype, 2022-03-10:
When ordering a new batch of PCB, there's always a few small modifications to be made, like minor errors, cosmetic mods, etc. This can be one of those, and may simply be a layer change from silkscreen to fab.
I usually create a new branch in Git when I order a PCB, with the version number as its name, to markdown the design at this point. I also use this commit to specify which changes were made.