yaqwsx/KiKit #307: Export STL files error

dj140, 2022-02-09:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

kikit, version 1.0.2

### KiCAD version

6.0.1

### Operating system

win10

### Description

E:\FurutaPendulum_PJT\Motor_Board>kikit stencil createprinted --pcbthickness 1.6 STM32_FOC.kicad_pcb ./

Traceback (most recent call last):
  File "E:\6.0\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "E:\6.0\bin\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "E:\6.0\bin\Scripts\kikit.exe\__main__.py", line 7, in <module>
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "E:\6.0\bin\Lib\site-packages\kikit\stencil_ui.py", line 26, in createPrinted
    return stencil.createPrinted(**kwargs)
  File "E:\6.0\bin\Lib\site-packages\kikit\stencil.py", line 473, in createPrinted
    renderScad(bottomStencilFile, os.path.join(outputdir, "bottomStencil.stl"))
  File "E:\6.0\bin\Lib\site-packages\kikit\stencil.py", line 282, in renderScad
    subprocess.check_call(["openscad", "-o", outfile, infile])
  File "E:\6.0\bin\Lib\subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['openscad', '-o', 'E:\\FurutaPendulum_PJT\\Motor_Board\\bottomStencil.stl', 'E:\\FurutaPendulum_PJT\\Motor_Board\\bottomStencil.scad']' returned non-zero exit status 1.

### Steps to Reproduce

kikit stencil createprinted --pcbthickness 1.6 STM32_FOC.kicad_pcb ./

yaqwsx, 2022-02-09:
Are you sure to have all the optional dependencies installed? I think you miss OpenSCAD - you have to be able to invoke it from CMD via calling `openscad`.

The error message should be, though, improved.

dj140, 2022-02-10:
I install openscad, it can invoke from CMD via `openscad`, now the error message change.. 

```
E:\FurutaPendulum_PJT\Motor_Board>kikit stencil createprinted --pcbthickness 1.6 STM32_FOC.kicad_pcb ./
\\\\\\\\\WARNING: Can't open DXF file 'E:/FurutaPendulum_PJT/Motor_Board\E:FurutaPendulum_PJTMotor_BoardSTM32_FOC-EdgeCuts.dxf'.
WARNING: Can't open DXF file 'E:/FurutaPendulum_PJT/Motor_Board\E:FurutaPendulum_PJTMotor_BoardSTM32_FOC-PasteBottom.dxf'.
Geometries in cache: 6
Geometry cache size in bytes: 96
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.000
Current top level object is empty.
Traceback (most recent call last):
  File "E:\6.0\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "E:\6.0\bin\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "E:\6.0\bin\Scripts\kikit.exe\__main__.py", line 7, in <module>
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "E:\6.0\bin\Lib\site-packages\kikit\stencil_ui.py", line 26, in createPrinted
    return stencil.createPrinted(**kwargs)
  File "E:\6.0\bin\Lib\site-packages\kikit\stencil.py", line 473, in createPrinted
    renderScad(bottomStencilFile, os.path.join(outputdir, "bottomStencil.stl"))
  File "E:\6.0\bin\Lib\site-packages\kikit\stencil.py", line 282, in renderScad
    subprocess.check_call(["openscad", "-o", outfile, infile])
  File "E:\6.0\bin\Lib\subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['openscad', '-o', 'E:\\FurutaPendulum_PJT\\Motor_Board\\bottomStencil.stl', 'E:\\FurutaPendulum_PJT\\Motor_Board\\bottomStencil.scad']' returned non-zero exit status 1.
```

dj140, 2022-02-10:
> The error message was improved in [fcbb7db](https://github.com/yaqwsx/KiKit/commit/fcbb7db0d54603aeb4330e44fc5fa804258aa7e1). In the case that this is not caused by OpenSCAD missing, could you please share the board? Otherwise, I cannot reproduce the issue and, therefore, fix the bug.

After changing the stencil.py according to [fcbb7db](https://github.com/yaqwsx/KiKit/commit/fcbb7db0d54603aeb4330e44fc5fa804258aa7e1), the error message become
```
E:\FurutaPendulum_PJT\Motor_Board>kikit stencil createprinted --pcbthickness 1.6 STM32_FOC.kicad_pcb ./
Traceback (most recent call last):
  File "E:\6.0\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "E:\6.0\bin\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "E:\6.0\bin\Scripts\kikit.exe\__main__.py", line 7, in <module>
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "E:\6.0\bin\Lib\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "E:\6.0\bin\Lib\site-packages\kikit\stencil_ui.py", line 25, in createPrinted
    from kikit import stencil
  File "E:\6.0\bin\Lib\site-packages\kikit\stencil.py", line 282
    try:
IndentationError: unexpected indent
```

yaqwsx, 2022-02-10:
It seems that OpenSCAD on Windows doesn't handle absolute paths in imports. I will reach out to the OpenSCAD maintainers and try to find a solution.

yaqwsx, 2022-02-10:
OpenSCAD requires to use forward slashes in imports instead of backslashes. Fixed in [5d1bbe9](https://github.com/yaqwsx/KiKit/commit/5d1bbe9873f068aee428e1106b542a60c6539f08)

closed 2022-02-10 by yaqwsx