yaqwsx/KiKit #308: Focus lost in Panelize

dzarda, 2022-02-11:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

1.0.2

### KiCAD version

6.0.1

### Operating system

Windows 10

### Description

Hi! Clicking Panelize in the form causes the window focus to be lost somehow and the user is then forced to navigate back to the Kicad window.

Please observe here
[image]

### Steps to Reproduce

- Press Panelize on Windows

dzarda, 2022-02-11:
Note the operation succeeds, the panel is generated :)

yaqwsx, 2022-02-12:
This was fixed in [7f6ace7](https://github.com/yaqwsx/KiKit/commit/7f6ace7c946d95bb9e2e230fc4c22bacfda34ef0)

closed 2022-02-12 by yaqwsx