yaqwsx/KiKit #309: Can't install kikit

mtjohnson47, 2022-02-13:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

'kikat' is not recognized as an internal or external command, operable program or batch file.

### KiCAD version

6.0.2

### Operating system

Windows

### Description

I cannot install kikit on a fresh kicad installation.

I get the following error:
`ERROR: Could not find a version that satisfies the requirement kikat (from versions: none)`
`ERROR: No matching distribution found for kikat`

Python version is Python 3.9.10.

### Steps to Reproduce

Install KiCad 6.0.2
Open the KiCad 6.0 command prompt as an administrator
typep "pip install kicad"

yaqwsx, 2022-02-13:
The tool name is 'kikit' not 'kikat' nor 'kicad'. I guess you just have misspelled the name, thus the error.

yaqwsx, 2022-03-01:
Closing due to inactivity. If the problem persists, please feel free to reopen.

closed 2022-03-01 by yaqwsx