yaqwsx/KiKit #311: Infinite overhang tolerance

dzarda, 2022-02-14:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

Hi! It appears to me the the "input overhang tolerance" should by default be set to a much larger value or "infinity". The confusion with my-components-are-gone will never go away and the RTFM argument is not good with such a critical behaviour that the user encounters immediately upon using the tool for the first time.

yaqwsx, 2022-02-15:
What would be a suitable value according to you?

dzarda, 2022-02-17:
I would prefer that by default the entire content of the board be duplicated verbatim to the panel. (999km tolerance)

yaqwsx, 2022-04-27:
A better approach: Consider only origin for footprints.

jj-uk, 2022-05-03:
Maybe another layer with a specific name could be used to indicate the outline to be used for panelising?

yaqwsx, 2022-05-03:
That is definitely the solution for some delicate cases; however, I don't want to force users to draw it for simple designs (most of the use-cases). This is why there is the automatic detection of the board, and, why why are looking for more sensible defaults on the auto-detection behavior.

jj-uk, 2022-05-03:
I think infinite should be the default. This would stop the "where are my components?" issues being raised.
If a smaller bounding area is needed because a user needs to add stuff to production layers for some reason, tolerance could be set to >0 in those cases.

I found this issue originally as a new user that was about to report a bug as some of my components that hang over the board we removed.

yaqwsx, 2022-05-03:
Infinite by default messes up multi-board workflow. This is the main reason why I won't implement it. I really like the idea of specifying the board by its outline (the current way); we just need a better way to decide what fits inside the board. IMHO "Consider only origin for footprints, whole graphics for the rest" for the inclusion test is the best approach at the moment.

dzarda, 2022-05-03:
The origin approach seems appropriate.

yaqwsx, 2022-05-12:
Implemented in d386887

closed 2022-05-12 by yaqwsx