yaqwsx/KiKit #312: Having trouble installing on Kubuntu 21.10

PigeonFX, 2022-02-14:
Follow the instructions but when trying to bring up the help menu I get the following:

```
$ kikit --help
Command 'kikit' not found, did you mean:
  command 'kinit' from deb heimdal-clients (7.7.0+dfsg-2)
  command 'kinit' from deb krb5-user (1.18.3-5)
```

Seems to have installed as if I try again to install I get:
```
$ pip3 install kikit
Requirement already satisfied: kikit in ./.local/lib/python3.9/site-packages (1.0.2)
Requirement already satisfied: pybars3>=0.9 in ./.local/lib/python3.9/site-packages (from kikit) (0.9.7)
Requirement already satisfied: shapely>=1.7 in ./.local/lib/python3.9/site-packages (from kikit) (1.8.0)
Requirement already satisfied: numpy>=1.21 in ./.local/lib/python3.9/site-packages (from kikit) (1.22.2)
Requirement already satisfied: commentjson>=0.9 in ./.local/lib/python3.9/site-packages (from kikit) (0.9.0)
Requirement already satisfied: solidpython>=1.1.2 in ./.local/lib/python3.9/site-packages (from kikit) (1.1.2)
Requirement already satisfied: pcbnewTransition==0.2.0 in ./.local/lib/python3.9/site-packages (from kikit) (0.2.0)
Requirement already satisfied: markdown2>=2.4 in ./.local/lib/python3.9/site-packages (from kikit) (2.4.2)
Requirement already satisfied: click>=7.1 in /usr/lib/python3/dist-packages (from kikit) (7.1.2)
Requirement already satisfied: lark-parser<0.8.0,>=0.7.1 in ./.local/lib/python3.9/site-packages (from commentjson>=0.9->kikit) (0.7.8)
Requirement already satisfied: PyMeta3>=0.5.1 in ./.local/lib/python3.9/site-packages (from pybars3>=0.9->kikit) (0.5.1)
Requirement already satisfied: euclid3<0.2.0,>=0.1.0 in ./.local/lib/python3.9/site-packages (from solidpython>=1.1.2->kikit) (0.1)
Requirement already satisfied: pypng<0.0.20,>=0.0.19 in ./.local/lib/python3.9/site-packages (from solidpython>=1.1.2->kikit) (0.0.19)
Requirement already satisfied: ply<4.0,>=3.11 in ./.local/lib/python3.9/site-packages (from solidpython>=1.1.2->kikit) (3.11)
Requirement already satisfied: PrettyTable==0.7.2 in ./.local/lib/python3.9/site-packages (from solidpython>=1.1.2->kikit) (0.7.2)

```

yaqwsx, 2022-02-14:
You probably don't have local Python packages in PATH. See https://stackoverflow.com/a/59436732/3248821

PigeonFX, 2022-02-14:
Thank you so much, that fixed the issue! 

Sorry about posting in the wrong place.

closed 2022-02-14 by PigeonFX