yaqwsx/KiKit #313: KiKit make v-cut over a PCB and not make backbone

ifadiga, 2022-02-16:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

1.0.2.

### KiCAD version

6.0.2.

### Operating system

Windows 10

### Description

When I use backbone and v-cut, KiKit doesn't make backbone between all boards. In the picture, you can see that backbone isn't made in the first and last row between the second and third PCB in these rows ( you can see a yellow rectangle where the backbone is missed), 

Also when I  use v-cut KiKit puts it over the first row of PCB (you can see on picture blue rectangle) and doesn't make a v-cut between the frame and PCBs ( I marked the place where need v-cut, between PCBs and frame, with a blue line). 
On GUI for v-cut, you added a parameter for offset but it doesn't change anything (for mouse bytes it works as expected, but doesn't work for v-cut).

Is possible to save my settings when making panels and use them on other PCBs or do I need to put all parameters every time?

[image]

### Steps to Reproduce

CLI command:
```
kikit panelize \
    --layout 'alternation: rowsCols; hspace: 0.2mm; vspace: 0.2mm; hbackbone: 0.2mm; vbackbone: 0.2mm; rows: 5; cols: 3' \
    --tabs 'type: none' \
    --cuts 'type: vcuts; layer: Eco1.User' \
    --framing 'type: railstb; vspace: 0mm; width: 6mm' \
    --tooling 'type: 4hole; hoffset: 4mm; voffset: 4mm; size: 2mm; paste: True' \
    --fiducials 'type: 4fid; hoffset: 8mm; voffset: 4mm; opening: 1.5mm' \
    --text 'type: simple; voffset: 2mm; text: RTC PANEL' \
    C:\Users\Ivan\Desktop\RTC_TEST_PCB KiCad6\RTC_PCF85063A.kicad_pcb panel.kicad_pcb

```
KiKit JSON preset:
```
{
    "layout": {
        "alternation": "rowsCols",
        "hspace": "0.2mm",
        "vspace": "0.2mm",
        "hbackbone": "0.2mm",
        "vbackbone": "0.2mm",
        "rows": "5",
        "cols": "3"
    },
    "tabs": {
        "type": "none"
    },
    "cuts": {
        "type": "vcuts",
        "layer": "Eco1.User"
    },
    "framing": {
        "type": "railstb",
        "vspace": "0mm",
        "width": "6mm"
    },
    "tooling": {
        "type": "4hole",
        "hoffset": "4mm",
        "voffset": "4mm",
        "size": "2mm",
        "paste": "True"
    },
    "fiducials": {
        "type": "4fid",
        "hoffset": "8mm",
        "voffset": "4mm",
        "opening": "1.5mm"
    },
    "text": {
        "type": "simple",
        "voffset": "2mm",
        "text": "RTC PANEL"
    }
}
```

yaqwsx, 2022-02-19:
What are you trying to achieve? Please describe your panel. You specify no tabs and you specify backbone. There are no cuts as cuts are only procured by tabs (and several other features like backbone cuts).

What you probably want to use is no backbone and tabs style `full`. When you use V-cuts, you probably also want 0 spacing. Otherwise, there will be 2 V-cuts 0.2mm apart which the fab house probably won't make for you.

The offsets for V-cuts are not implemented (#215) and it is a bug that GUI allows you to specify it. It is ignored at the moment.

If you want to apply the same settings for multiple boards, just save the JSON or KiCAD command and run the tool from CLI.

ifadiga, 2022-02-23:
Hello,
Sorry for I didn't understand how KiKit works. I tried to use tabs as you explained and removed spacing because, if I have a little spacing KiKit made 2 V-cuts and that will be a problem for the fab house as you say (I saw 2 V-cuts after reading your answer).

After changing all parameters I got the panel as I want. Sorry, one more time, and thanks for helping with my problems.

Also for saving I copied the command with all settings and use it with CLI as you suggested.

closed 2022-02-23 by ifadiga