yaqwsx/KiKit #316: SLA printer make PCB

hedon77, 2022-02-20:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

Hi,

Thanks for the plugin. Could you add the ability to generate STL from any kicad layer. I would like to make single-sided PCBs as in the movie.

Greetings 

https://www.youtube.com/watch?v=-Qeq7ZgUOuE

yaqwsx, 2022-02-20:
You can already do so with https://github.com/sn4k3/UVtools - simply export your copper layer into a bitmap and then import it as layers. I think it is fewer steps than exporting STL from KiKit and then loading it into a slicer. I do not plan to support the automatic generation of sliced files in KiKit due to the large palette of formats out there.

yaqwsx, 2022-07-05:
Since new printers & encrypted formats are coming, I decided to not implement this feature in KiKit. Instead, you can directly use UVTools that can render gerber files and prepare a file for "printing".

[image]
[image]

closed 2022-07-05 by yaqwsx