yaqwsx/KiKit #320: len(symbols) is not equal to len(instances)

FrankvVeelen, 2022-02-25:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

1.0.2

### KiCAD version

6.0.2

### Operating system

Ubuntu

### Description

When running kikit fab jlcpcb project_name.kicad_pcb fabrication/ --assembly --schematic project_name.kicad_sch --no-drc, kikit crashes with the following output:

`
Traceback (most recent call last):
  File "/home/frank/.local/bin/kikit", line 8, in <module>
    sys.exit(cli())
  File "/home/frank/.local/lib/python3.8/site-packages/click/core.py", line 1137, in __call__
    return self.main(*args, **kwargs)
  File "/home/frank/.local/lib/python3.8/site-packages/click/core.py", line 1062, in main
    rv = self.invoke(ctx)
  File "/home/frank/.local/lib/python3.8/site-packages/click/core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/frank/.local/lib/python3.8/site-packages/click/core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/frank/.local/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/frank/.local/lib/python3.8/site-packages/click/core.py", line 763, in invoke
    return __callback(*args, **kwargs)
  File "/home/frank/.local/lib/python3.8/site-packages/kikit/fab_ui.py", line 32, in jlcpcb
    return jlcpcb.exportJlcpcb(**kwargs)
  File "/home/frank/.local/lib/python3.8/site-packages/kikit/fab/jlcpcb.py", line 73, in exportJlcpcb
    components = extractComponents(schematic)
  File "/home/frank/.local/lib/python3.8/site-packages/kikit/fab/common.py", line 19, in extractComponents
    return eeschema_v6.extractComponents(filename)
  File "/home/frank/.local/lib/python3.8/site-packages/kikit/eeschema_v6.py", line 152, in extractComponents
    assert len(symbols) == len(instances)
AssertionError
`

### Steps to Reproduce

Boards are confidential, and i can't figure out how to reproduce with a minimal example. I do use hierarchical sheets multiple times, as in one sheet has 3 instances, so that may have to with it. It's also a converted project from KicadV5 which may have something do with it as well.

yaqwsx, 2022-02-25:
If you want it is possible to share the board confidentially. If you are interested, please, contact me at email@honzamrazek.cz to discuss further details. Having the boards will allow me to find the bug quicker.

Based on the assert; is it possible that you use multi-unit symbols, possibly spread between multiple sheets?

FrankvVeelen, 2022-02-25:
I do indeed use multi-unit symbols, one of which is the CM4IO, among others. 
Weird thing is that the CM4IOv5 project seems to work fine (downloadable from [http://datasheets.raspberrypi.org/cm4io/CM4IO-KiCAD.zip](http://datasheets.raspberrypi.org/cm4io/CM4IO-KiCAD.zip)). Do you have any idea what the problem may be?

yaqwsx, 2022-02-28:
The problem is that there is no API for schematics so KiKit parses the schematics and interprets it. Therefore it is possible that v5 works fine and there is a bug in v6. 

I am happy that you were able to find out the problem!

closed 2022-02-28 by yaqwsx