yaqwsx/KiKit #321: Discontinuous line on multiple board PCB

troisiemetype, 2022-02-27:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

1.0.2

### KiCAD version

6.0.2

### Operating system

Lubuntu

### Description

I'm currently designing a multi-board circuit with four boards, two of them have to be made twice (for a total of six boards). Multi board design went well with the help of your documentation. I would like to design the two board for which I want two occurrence only one so they are exactly the same, then extract them to panelize them together.
Extract works well too.
But I cannot panelize them, I've got the error `Discontinuous line at []`, the point expressed being the junction of an arc and a segment.
Both CLI, that I first tried) and a python script gave the same problem.

I've read the different issues regarding "discontinuous line". I've opened the kicad_pcb file in a text editor, the start and end points are exactly the same. I understand that this kind of problem comes from the resolution at which kicad is working, and I suppose that the coordinate being stored in the file is not the one used internally (maybe re-computed using the center coordinate ?) There's no way I can add a segment by hand between to points that are virtually the same.

I've also tried to replace one of the contour outline by a basic rectangle, I have the same error on one corner.

I've tried to follow the traceback of the python script, and ended up changing the rounding precision of `roundPoint()` in substrate.py (I've then seen you advise someone to do exactly that in another issue), but no success.

### Steps to Reproduce

Here is the repository of this project, pushed with this behavior : https://github.com/troisiemetype/freeformclock
Run the script at `panelization/panelize.py`, which runs on the `freeform_clock.kicad_pcb` in the main folder.
In the CLI I've used the following command : `kikit panelize -p freeformclock_settings.json --source 'annotation; ref: main' ../freeform_clock.kicad_pcb freeformclock_panel.kicad_pcb`

troisiemetype, 2022-02-27:
Note : if I export the four boards to intermediate kicad_pcb, both the CLI and the python script work and are able to panelize, with one or several boards respectively.

yaqwsx, 2022-02-27:
You set 'tolerance' to 15mm which means that some of the geometry from the power board gets included. Since it is not complete (the vertical lines are not included) you get discountinous-outline error.

yaqwsx, 2022-02-27:
I improved the error message in 915bc87.

troisiemetype, 2022-02-28:
Hello, thank you very much for your answer. I can't believe I'm so dumb I didn't see it, event after having look at all the parameters...
I've copied the script from one other issue, and changed some things in it to suit me.
Now it works, almost. Still some Continuous issues, but this time it's an arc on arc, so back to kicad's problem. :)

Again, thank you for answering that quick !

yaqwsx, 2022-02-28:
You can connect two arcs with a short linear segment. However, if you use KiCAD 6 the endpoint snapping should work well.

closed 2022-02-28 by yaqwsx