yaqwsx/KiKit #335: KiKit causes KiCad to crash when trying to panelize

electron271, 2022-03-27:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

kikit, version 1.0.3

### KiCAD version

6.0.4

### Operating system

Windows 11

### Description

KiCad stops responding and crashes when trying to panelize through a GUI

### Steps to Reproduce

1. Open panelizing tool
2. Attempt to panelize a PCB with the default settings
3. Start the panelizing tool

electron271, 2022-03-27:
I did a test with a 1.65x1.90in PCB demo, (ecc83-pp_v2 KiCad demo), and it still crashes.

electron271, 2022-03-27:
I think I found the problem, Powershell doesn't support the character "\" for starting a new line, so it did nothing.

electron271, 2022-03-27:
Sort of followed in #336

closed 2022-03-27 by electron271

yaqwsx, 2022-03-28:
Hi, could you please share the source board? The backslashes cannot cause GUI to crash (as they are not used in the process). Is it possible that you tried to panelize your board when your board was already opened in the same window? This causes problems since KiCAD 6.0.4 ([related issue](https://gitlab.com/kicad/code/kicad/-/issues/10540)) (and I just added a check for that in 0ba8262)

electron271, 2022-03-28:
I am using the ecc83-pp_v2 KiCad demo, which is builtin to KiCad

yaqwsx, 2022-03-28:
Did you open the file and then in the same window tried to make panel where input was the same file?

electron271, 2022-03-28:
Ok, that seemed to be the problem, I tried while opening a different board, and it worked!