yaqwsx/KiKit #336: Change CLI command for Windows

electron271, 2022-03-27:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

Change the CLI command in the examples and GUI for windows to use backticks instead of backslashes to indicate new line.

electron271, 2022-03-27:
This issue will fix #335, a closed issue which had the original problem

yaqwsx, 2022-03-28:
I am not a Windows user, however, it seems that:
- PowerShell uses backticks
- Command Prompt (and thus also the KiCAD Command Prompt) use `^`

The proposed solution is to add a combo box to choose a delimiter for the command and select either backlash (when running on Linux) and  `^` when running on Windows. The backticks can be an option but I wouldn't use them as default since running KiKit in Powershell is not officially supported (as it requires extra setup and caution from the user).

yaqwsx, 2022-05-08:
This was addressed in f705d07 and 6b8c604215f87f068a45c206aff31cac2da85e0b

closed 2022-05-08 by yaqwsx