yaqwsx/KiKit #337: An error occurred: Unsupported shape 4 in outline

acruise, 2022-03-29:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

kikit, version 1.0.3+15.g1155a2d

### KiCAD version

6.0.4-1.fc35

### Operating system

Fedora 35

### Description

When I try to panelize any of my boards (they all originate from the same copy-paste), I get the same error:

```
root@45c68b123c8c:/kikit# kikit panelize 4pin/4pin.kicad_pcb 4pin/panel
herekikit.ui
An error occurred: Unsupported shape 4 in outline
No output files produced
```

This is running in docker on a Fedora 35 host, but I was seeing the same symptom in Windows. I can't figure out how to use `--debug` at all.

### Steps to Reproduce

1. Download https://drive.google.com/file/d/1ylKNAjH_UuK4lkmlOfECCkZKOf9BMU4H/view?usp=sharing
2. `kikit panelize 4pin.kicad_pcb panel`

yaqwsx, 2022-03-29:
At the moment, KiKit doesn't support polygons in outline. I will add this support, in the meantime as a workaround you can select the outline -> right-click -> convert to lines. Then the board should work fine with KiKit.

yaqwsx, 2022-03-29:
Support for polygons was implemented in 53a0e9b510e4eb069a215bd9d52cf9854393da48

closed 2022-03-29 by yaqwsx

acruise, 2022-03-29:
I tried converting the outer border to lines, but it had no effect. :/

yaqwsx, 2022-03-29:
Then it means you didn't convert them to lines - if you have successfully converted them to lines, there would be no polygon shape. Nevertheless, upgrade to KiKit 1.0.4 which I made today and your board should work just fine with no changes.

acruise, 2022-03-29:
yes, it turns out the conversion leaves the polygon in place, I had to delete it afterward. This looks great now!

btw, the checkerboard alternation pattern was also buggy in 1.0.3, only the second and fourth rows were alternating.

[image]