yaqwsx/KiKit #338: Doesn't work on bare metal Fedora 35, possibly dependency issue

acruise, 2022-03-29:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

kikit, version 1.0.3+15.g1155a2d

### KiCAD version

6.0.4-1.fc35

### Operating system

Fedora 35

### Description

When I `pip install kikit` then try to run it, I get this:

Maybe my version of `click` is too new?

```
[alex@leftylinux JST-XH breakouts]$ kikit
Traceback (most recent call last):
  File "/home/alex/.local/bin/kikit", line 5, in <module>
    from kikit.ui import cli
  File "/home/alex/.local/lib/python3.10/site-packages/kikit/ui.py", line 2, in <module>
    from kikit import (panelize_ui, export_ui, present_ui, stencil_ui,
  File "/home/alex/.local/lib/python3.10/site-packages/kikit/panelize_ui.py", line 148, in <module>
    def panelize(input, output, preset, layout, source, tabs, cuts, framing,
  File "/home/alex/.local/lib/python3.10/site-packages/click/decorators.py", line 316, in decorator
    _param_memo(f, OptionClass(param_decls, **option_attrs))
  File "/home/alex/.local/lib/python3.10/site-packages/click/core.py", line 2469, in __init__
    super().__init__(param_decls, type=type, multiple=multiple, **attrs)
TypeError: Parameter.__init__() got an unexpected keyword argument 'autocompletion'
```

### Steps to Reproduce

1. `pip install kikit` on Fedora 35
2. `kikit`

acruise, 2022-03-29:
Workaround: `pip install click==7.1` :)

yaqwsx, 2022-03-29:
Click 8.1 was released yesterday which removed the keyword argument `autocompletion`. I wasn't restricting the version of click on purpose, so we have the least collision with other packages. However, since 8.1 comes with breaking changes, I will adjust the dependencies.

yaqwsx, 2022-03-29:
Resolved by adefbf98aacd6394ef9afec6ca7bf1c8ab8bf6a8

closed 2022-03-29 by yaqwsx