yaqwsx/KiKit #339: KiKit installation help on Win10 KiCad 6.0.2

sujitth, 2022-03-29:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.0 (older version are not supported)

### KiKit version

V1.03

### KiCAD version

6.0.2

### Operating system

Windows10x64bit

### Description

Hi,

I have installed Kikit as per the installation instructions successfully, but when checking the command kikit --version after installation getting an error as attached below:

[image]

and in Pcbnew, only the Show/hide menu is displayed.

Please help to resolve the installation.

Thanks,
Sujith

### Steps to Reproduce

-NA-

yaqwsx, 2022-03-29:
This is a duplicate of #338.

Yesterday, a new version of the click library was released that is no longer backward compatible with the feature set we use. We didn't restrict the library version on purpose in the order to avoid collisions with other installed packages. The fix was already committed but wasn't released. As a temporary work-around, please downgrade Click to version 8.0.0

TLDR: before installing KiKit, install Click 8.0.0 (`pip install click==8.0.0`)

closed 2022-03-29 by yaqwsx

sujitth, 2022-03-29:
Thanks for the reply.

do I need to uninstall the current kikit installation before installing
click 8.0.0? if so what is the commando uninstall?

yaqwsx, 2022-03-29:
You don't need to uninstall KiKit.

yaqwsx, 2022-03-29:
I made the 1.0.4 release so the problem shouldn't occur anymore.