yaqwsx/KiKit #344: SMT placement issue when creating fab files for jlcpcb

bipsendk, 2022-04-15:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.4 (older version are not supported)

### KiKit version

1.0.5

### KiCAD version

6.0.4

### Operating system

Windows 10

### Description

I created a panel of a circular PCB.

So far, so good....

Then I tried to prepare files for jlcpcb.

I got some warnings regarding missing ordercodes for some components - but that was fixed in the BOM file afterwards....

The problem occur with SMT assembly and location of the components:

[image]

Now - then question is where things go wrong ? 

### Steps to Reproduce

Panelize:

kikit panelize \
    --layout 'hspace: 6mm; vspace: 6mm; hbackbone: 3mm; vbackbone: 3mm; rows: 3; cols: 3; vbonecut: False; hbonecut: False' \
    --tabs 'type: fixed; vwidth: 5mm; hwidth: 5mm' \
    --cuts 'type: mousebites' \
    --framing 'type: frame; hspace: 3mm; vspace: 3mm; width: 5mm; cuts: none' \
    --tooling 'type: 4hole; hoffset: 2.5mm; voffset: 2.5mm; size: 1.152mm' \
    --fiducials 'type: 4fid; hoffset: 5mm; voffset: 2.5mm; coppersize: 1.5mm' \
    --post 'millradius: 1mm' \
    S:\SmartHome\KamstrupIR\kamstrup.kicad_pcb panel.kicad_pcb

(remember to re-do zones)

kikit fab jlcpcb --assembly --schematic S:\SmartHome\KamstrupIR\kamstrup.kicad_sch   S:\SmartHome\KamstrupIR\Kamstrup_panel.kicad_pcb S:\SmartHome\KamstrupIR\PanelOut

[kamstrup-panel.zip](https://github.com/yaqwsx/KiKit/files/8495979/kamstrup-panel.zip)

Then try to add things to jlcpcb and select smt assembly and upload files....

closed 2022-04-21 by bipsendk

conradopoole, 2022-07-26:
@bipsendk What did you do to get around this issue? I'm facing a similar one, all my parts are offset, both top and left. When I run the kikit fab command on the single board PCB everything is placed correctly, but not when I run it on a panel.

conradopoole, 2022-07-26:
[image]
@yaqwsx I've tried with different values for the Origin in the "post" section but none get the right placement part, they get better or worse but none get it right. I wonder if it does have something to do with origins and how kikit is calculating the PosX and PosY for all components in the panel.

bipsendk, 2022-07-27:
@conradopoole I used jlc-kicad-tools to handle the positioning of components...

yaqwsx, 2022-08-17:
Hi, this problem was fixed by [1a23ce3](https://github.com/yaqwsx/KiKit/commit/1a23ce3324dff7728296eb55730b114f0e1db5fd).

When generating gerbers for JLC PCB, we didn't consider auxiliary origin. This was fine until we implemented panel positioning (6458178bfe8123daf5c11de775f89a10d9570469) that shifts the auxiliary origin.