yaqwsx/KiKit #350: Tuple index out of range

ghilintadaniel, 2022-05-10:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.4 (older version are not supported)

### KiKit version

kikit, version 1.0.5

### KiCAD version

6.0.1, release build

### Operating system

Windows 10

### Description

The plugin gives "Tuple index out of range" error, from Python. Here is the traceback:
Traceback (most recent call last):
  File "C:\Users\ghili\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\panelize_ui.py", line 182, in panelize
    doPanelization(input, output, preset)
  File "C:\Users\ghili\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\panelize_ui.py", line 222, in doPanelization
    tabCuts = ki.buildTabs(preset["tabs"], panel, substrates,
  File "C:\Users\ghili\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\panelize_ui_impl.py", line 262, in buildTabs
    return panel.buildFullTabs(frameOffsets)
  File "C:\Users\ghili\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\panelize.py", line 1287, in buildFullTabs
    outerBounds = shpBBoxMerge(outerBounds, s.partitionLine.bounds)
  File "C:\Users\ghili\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\common.py", line 197, in shpBBoxMerge
    min(a[0], b[0]),
IndexError: tuple index out of range

### Steps to Reproduce

I think the problem comes from my pcb, which has a round cut on the edges, like this
[image]
.

I am 99% sure that the line is continuous. 

The CLI command used is the next one, followed by the path to pcb file:
kikit panelize \
    --layout 'rows: 2; cols: 2' \
    --source 'tolerance: 20mm' \
    --tabs 'type: full' \
    --cuts 'type: vcuts' \
    --framing 'type: railslr; hspace: 0mm; width: 7mm' \
    --tooling 'type: 4hole; hoffset: 3.5mm; voffset: 10mm; size: 2mm' \
    --fiducials 'type: 4fid; hoffset: 3.5mm; voffset: 5mm' \

yaqwsx, 2022-05-10:
Could you share the board outline? Without it will be hard to debug - since this is a bug in full tabs implementation. 

Nevertheless, based on your outline I think you don't want to use full tabs as full tabs will remove your notch on the right side. You probably want to place a large tab in the middle.

ghilintadaniel, 2022-05-10:
Attached are the board outline and a printscreen with the desired panel. The panel could be 2x2 or 1x3.
[kikit_error.zip](https://github.com/yaqwsx/KiKit/files/8663365/kikit_error.zip)
.

yaqwsx, 2022-05-10:
Well, I was wrong. I have to give a more thought.

yaqwsx, 2022-05-11:
Closed via da0bb0b.

closed 2022-05-11 by yaqwsx