yaqwsx/KiKit #353: Different tab spacing values for horizontal and vertical

electroniceel, 2022-05-13:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

I have used these settings:
```
{
    "layout": {
        "hspace": "2.5mm",
        "vspace": "2.5mm",
        "rows": "5"
    },
    "tabs": {
        "vwidth": "4mm",
        "hwidth": "4mm",
        "spacing": "15mm"
    },
    "framing": {
        "type": "railslr",
        "width": "7mm",
        "chamfer": "2mm"
    },
    "post": {
        "millradius": "1mm"
    }
}
```
and Kikit creates this panel out of it:

[image]

The 15mm spacing for the tabs is necessary to let Kikit produce proper tabs on the left and right. But this number of tabs on the upper/lower side between the pcbs is excessive. I would prefer to use 50mm spacing there. But when I configure it, the tabs on the left and right vanish:

[image]

So it would be nice to be able to configure different values for horizontal and vertical tabs. But I'm of course also open for better solutions for my problem.

yaqwsx, 2022-05-13:
You can already do so - see fixed tab style in [documentation](https://github.com/yaqwsx/KiKit/blob/master/doc/panelizeCli.md#tabs). If you need a precise tab placement, you can use annotations and once v1.1 will be released we will support third-party plugins for tab placement.

electroniceel, 2022-05-13:
You are right, with the fixed tab style it works. Thanks for your help.

closed 2022-05-13 by electroniceel