yaqwsx/KiKit #355: CLI comands in Windows, ' instead of "

androot2, 2022-05-17:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.4 (older version are not supported)

### KiKit version

1.0.5

### KiCAD version

6.0.4

### Operating system

Windows 10

### Description

I tried kikit run from command line, just copy that:
[image]
but in Windows is neccesary to replace ' by ". Instead of this I got an error:
[image]
Maybe this is the reason why I cannot create panel from kikit GUI?

Other issue are extra <cr><lf>, which should be removed before paste in command window.

### Steps to Reproduce

I tried copy commands from GUI window to Kicad command prompt, results are above.

yaqwsx, 2022-05-17:
This was already addressed in 6b8c604215f87f068a45c206aff31cac2da85e0b. It will be part of the upcoming release. 

The GUI doesn't invoke the command, instead it calls internal functions directly so it is unlikely that this is the cause.

yaqwsx, 2022-06-29:
Is this resolved or is there anything else that needs to be addressed?

closed 2022-08-08 by yaqwsx