yaqwsx/KiKit #356: [Kikit Docker] DRC throws error "Unable to access the X Display, is $DISPLAY set properly?"

duckyb, 2022-05-18:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.4 (older version are not supported)

### KiKit version

latest

### KiCAD version

6

### Operating system

ubuntu-latest (docker)

### Description

Yesterday I was trying to run DRC on my project but I ran into issue #270 . Switching to Kikit `nightly` resulted the error below, while `latest` had the #270 bug.

Today, since `latest` has been updated it also has this new bug.

### Steps to Reproduce

1. Run a new docker image using kikit:latest
2. run `kikit drc run` on any `.kicad_pcb` file

### GitHub Action:
```
name: 'Run DRC'
description: 'Check design rules on pcb'
inputs:
  pcb_file:
    description: 'Kicad input file'
    required: true
  flags:
    description: 'Additional flags for kikit drc'
    default: ''
runs:
  using: 'docker'
  image: 'docker://yaqwsx/kikit:latest'
  entrypoint: '/bin/sh'
  args:
    - '-c'
    - kikit drc run ${{ inputs.flags }} $GITHUB_WORKSPACE/${{ inputs.pcb_file }}
```
**output (shortened):**
```
kikit drc run pcb.kicad_pcb
Unable to find image 'yaqwsx/kikit:latest' locally
latest: Pulling from yaqwsx/kikit

Status: Downloaded newer image for yaqwsx/kikit:latest
Unable to access the X Display, is $DISPLAY set properly?
```

yaqwsx, 2022-05-18:
Fixed in 96b0030, the change was backported to the 1.0.5 and latest images.

closed 2022-05-18 by yaqwsx

duckyb, 2022-05-18:
Thank you for the fix 👍🏻