yaqwsx/KiKit #363: macOS install instructions not working

elmegil, 2022-05-29:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.4 (older version are not supported)

### KiKit version

kikit: command not found (trying to install from latest github)

### KiCAD version

6.0.5-0

### Operating system

MacOS 10.15.7 (Catalina)

### Description

Running the install process as described in https://github.com/yaqwsx/KiKit/blob/master/doc/macosInstallation.md

I have created the certificate
[image]

But after running sudo bash scripts/installMacOS.bash

final output is

KiCAD will be resigned
error: The specified item could not be found in the keychain.

Poking at this some with notes from stack exchange etc I see:

Pete-MacBook-Pro:KiKit petehartman$ security find-identity

Policy: X.509 Basic
  Matching identities
  1) 75C48942868E51742AE4A641C9689D4E2C09EA65 "com.apple.idms.appleid.prd.7a31315274467a6e5555522b77324e6a5278505952673d3d" (CSSMERR_TP_CERT_EXPIRED)
  2) 87E4D2F504F8DC8412C6A357E81E804FBF213B16 "com.apple.idms.appleid.prd.7a31315274467a6e5555522b77324e6a5278505952673d3d" (CSSMERR_TP_CERT_EXPIRED)
  3) C70B554731A08EC98C598C7386F5C24AF12D754C "com.apple.idms.appleid.prd.7a31315274467a6e5555522b77324e6a5278505952673d3d" (CSSMERR_TP_CERT_EXPIRED)
  4) FA886B695C4C1A1B34022F0CEA1B8CB6AA7BF138 "com.apple.idms.appleid.prd.7a31315274467a6e5555522b77324e6a5278505952673d3d" (CSSMERR_TP_CERT_EXPIRED)
  5) 98E08A773119C5C5249479F8B2AFD9D7C425C75E "com.apple.idms.appleid.prd.7a31315274467a6e5555522b77324e6a5278505952673d3d" (CSSMERR_TP_CERT_EXPIRED)
  6) 155078EA4C03E14C79A7DA41B110B500DA4DE28C "com.apple.idms.appleid.prd.7a31315274467a6e5555522b77324e6a5278505952673d3d" (CSSMERR_TP_CERT_REVOKED)
  7) E1F989C66421397F601ED7BD7F5F3E5DF8C08F86 "kikit" (CSSMERR_TP_NOT_TRUSTED)
     7 identities found

  Valid identities only
     0 valid identities found

I have so far been unable to find a way to make my certificate a trusted identity.

### Steps to Reproduce

Attempt to install kikit with "sudo bash scripts/installMacOS.bash" on Catalina, with a freshly created kikit certificate.

elmegil, 2022-05-29:
I have tried to see if I can create the cert somewhere other than under "logins" but the certificate assistant doesn't give any option about that

yaqwsx, 2022-05-29:
You have to create code-signing certificate.

elmegil, 2022-05-29:
I *did*

Just to be sure I have just now gone through, deleted the original, and tried again:

[image]
[image]
[image]
[image]
[image]

yaqwsx, 2022-06-01:
I am sorry; I missed that you already create a code-signing certificate on my phone.

I donẗ have a Mac to try and I haven't found anything useful by googling. Maybe someone from #252 could help?

tjlusco, 2022-06-16:
While not directly related to the code signing issue, the "installMacOS.bash" won't work as is. The heredoc isn't working, it just creates an empty file at "/usr/local/bin/kikit". [Remving the '|'  on this line](https://github.com/yaqwsx/KiKit/blob/d52b7c946fd80efbe4897f5223e1828eb6d5b92a/scripts/installMacOS.bash#L25), changing it to `cat << EOF > /usr/local/bin/kikit` fixes it.

On a side note, it also appears that running the script as "sudo" is unnecessary. "codesign" still works when run as the user, and all directories are owned by the user. I've successfully run the script without "sudo" after commenting out the check.

Finally, it might be worth [pointing the python version](https://github.com/yaqwsx/KiKit/blob/d52b7c946fd80efbe4897f5223e1828eb6d5b92a/scripts/installMacOS.bash#L9) to "Current", which symlinks back to 3.8 currently. This may only be in the nightlies, I don't have older versions installed to verify.

yaqwsx, 2022-06-29:
@tjlusco: If can fix it and validate it works, please, submit a PR. I don't have Mac available, so I am not able to make the changes by myself.

yaqwsx, 2022-08-17:
This was addressed by #385.

closed 2022-08-17 by yaqwsx