yaqwsx/KiKit #365: Changing 'page' causes error for anything but 'inherit'

a7hybnj2, 2022-06-02:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.4 (older version are not supported)

### KiKit version

1.0.5+59.g3fb0c3c

### KiCAD version

Application: KiCad PCB Editor  Version: (6.0.5-0)

### Operating system

macOS 12.3.1

### Description

Changing the 'page' field to anything !inherit cause a `unexpected 'end of input'... line 1, offset 1.`

### Steps to Reproduce

Load current and pcbnew plugin. Change page to USletter or anything and click panelize.

a7hybnj2, 2022-06-02:
Just rolled back to 1.0.5 and the page functionally works again.

yaqwsx, 2022-06-03:
This was caused by a recent optimization; I haven't covered it in tests, so it ended up buggy. Fixed in ba8c151fa93dded8f55eee2cdf76d8d921f5504e

a7hybnj2, 2022-06-03:
Yes, I just updated to current master and it appears to solve both problems. The error and the multiline text. Thanks!

closed 2022-06-03 by a7hybnj2