yaqwsx/KiKit #370: KiKit Installation error

greenscreenflicker, 2022-06-13:
Hello, I get the following error when installing:

C:\Users\mchhe\OneDrive\Dokumente\git\leddvr10w\FlybackV4_Hutschiene> pip3 install kikit
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: kikit in c:\users\mchhe\documents\kicad\6.0.5\3rdparty\python39\site-packages (1.0.5)
Requirement already satisfied: pcbnewTransition==0.2.0 in c:\users\mchhe\documents\kicad\6.0.5\3rdparty\python39\site-packages (from kikit) (0.2.0)
Requirement already satisfied: shapely>=1.7 in c:\users\mchhe\documents\kicad\6.0.5\3rdparty\python39\site-packages (from kikit) (1.8.2)
Requirement already satisfied: markdown2>=2.4 in c:\users\mchhe\documents\kicad\6.0.5\3rdparty\python39\site-packages (from kikit) (2.4.3)
Requirement already satisfied: commentjson>=0.9 in c:\users\mchhe\documents\kicad\6.0.5\3rdparty\python39\site-packages (from kikit) (0.9.0)
Requirement already satisfied: pybars3>=0.9 in c:\users\mchhe\documents\kicad\6.0.5\3rdparty\python39\site-packages (from kikit) (0.9.7)
Requirement already satisfied: click>=7.1 in c:\users\mchhe\documents\kicad\6.0.5\3rdparty\python39\site-packages (from kikit) (8.1.3)
Requirement already satisfied: solidpython>=1.1.2 in c:\users\mchhe\documents\kicad\6.0.5\3rdparty\python39\site-packages (from kikit) (1.1.3)
Requirement already satisfied: numpy>=1.21 in c:\program files\kicad\6.0\bin\lib\site-packages (from kikit) (1.22.3)
Requirement already satisfied: colorama in c:\users\mchhe\documents\kicad\6.0.5\3rdparty\python39\site-packages (from click>=7.1->kikit) (0.4.4)
Requirement already satisfied: lark-parser<0.8.0,>=0.7.1 in c:\users\mchhe\documents\kicad\6.0.5\3rdparty\python39\site-packages (from commentjson>=0.9->kikit) (0.7.8)
Requirement already satisfied: PyMeta3>=0.5.1 in c:\users\mchhe\documents\kicad\6.0.5\3rdparty\python39\site-packages (from pybars3>=0.9->kikit) (0.5.1)
Requirement already satisfied: euclid3<0.2.0,>=0.1.0 in c:\users\mchhe\documents\kicad\6.0.5\3rdparty\python39\site-packages (from solidpython>=1.1.2->kikit) (0.1)
Requirement already satisfied: ply<4.0,>=3.11 in c:\users\mchhe\documents\kicad\6.0.5\3rdparty\python39\site-packages (from solidpython>=1.1.2->kikit) (3.11)
Requirement already satisfied: pypng<0.0.20,>=0.0.19 in c:\users\mchhe\documents\kicad\6.0.5\3rdparty\python39\site-packages (from solidpython>=1.1.2->kikit) (0.0.19)
Requirement already satisfied: PrettyTable==0.7.2 in c:\users\mchhe\documents\kicad\6.0.5\3rdparty\python39\site-packages (from solidpython>=1.1.2->kikit) (0.7.2)
WARNING: There was an error checking the latest version of pip.

C:\Users\mchhe\OneDrive\Dokumente\git\leddvr10w\FlybackV4_Hutschiene>

Any proposals how to fix it?

greenscreenflicker, 2022-06-17:
Ping @yaqwsx

craigarno, 2022-06-24:
Looks like the problem is solved here https://forum.kicad.info/t/solved-kitkit-plugin-issue-installed-but-not-present/32867

_Earlier Warning_:
WARNING: You are using pip version 21.2.4; however, version 22.1.2 is available.
You should consider upgrading via the 'C:\Program Files\Python39\python.exe -m pip install --upgrade pip' command.

I upgraded pip as earlier warnings suggested, after opening the KiKad 6.0.6 Command Prompt "Run as Administrator"
```
C:\Users\craig\Documents\KiCad\6.0.6>pip install --upgrade pip
Requirement already satisfied: pip in c:\program files\kicad\6.0\bin\lib\site-packages (21.2.4)
Collecting pip
  Using cached pip-22.1.2-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 21.2.4
    Uninstalling pip-21.2.4:
      Successfully uninstalled pip-21.2.4
Successfully installed pip-22.1.2

C:\Users\craig\Documents\KiCad\6.0.6>pip --version
pip 22.1.2 from C:\Program Files\KiCad\6.0\bin\Lib\site-packages\pip (python 3.9)

C:\Users\craig\Documents\KiCad\6.0.6>pip install kikit
Collecting kikit
  Downloading KiKit-1.0.5-py3-none-any.whl (212 kB)
     ---------------------------------------- 212.8/212.8 kB 1.6 MB/s eta 0:00:00
Requirement already satisfied: numpy>=1.21 in c:\program files\kicad\6.0\bin\lib\site-packages (from kikit) (1.22.4)
Collecting pybars3>=0.9
  Using cached pybars3-0.9.7-py3-none-any.whl
Collecting commentjson>=0.9
  Using cached commentjson-0.9.0-py3-none-any.whl
Collecting markdown2>=2.4
  Downloading markdown2-2.4.3-py2.py3-none-any.whl (35 kB)
Collecting pcbnewTransition==0.2.0
  Using cached pcbnewTransition-0.2.0-py3-none-any.whl (4.3 kB)
Collecting solidpython>=1.1.2
  Using cached solidpython-1.1.3-py3-none-any.whl (2.7 MB)
Collecting shapely>=1.7
  Downloading Shapely-1.8.2-cp39-cp39-win_amd64.whl (1.3 MB)
     ---------------------------------------- 1.3/1.3 MB 3.4 MB/s eta 0:00:00
Collecting click>=7.1
  Downloading click-8.1.3-py3-none-any.whl (96 kB)
     ---------------------------------------- 96.6/96.6 kB 5.8 MB/s eta 0:00:00
Collecting colorama
  Downloading colorama-0.4.5-py2.py3-none-any.whl (16 kB)
Collecting lark-parser<0.8.0,>=0.7.1
  Using cached lark_parser-0.7.8-py2.py3-none-any.whl
Collecting PyMeta3>=0.5.1
  Using cached PyMeta3-0.5.1-py3-none-any.whl
Collecting ply<4.0,>=3.11
  Using cached ply-3.11-py2.py3-none-any.whl (49 kB)
Collecting euclid3<0.2.0,>=0.1.0
  Using cached euclid3-0.1-py3-none-any.whl
Collecting pypng<0.0.20,>=0.0.19
  Using cached pypng-0.0.19-py3-none-any.whl
Collecting PrettyTable==0.7.2
  Using cached prettytable-0.7.2-py3-none-any.whl
Installing collected packages: pypng, PyMeta3, PrettyTable, ply, lark-parser, euclid3, solidpython, shapely, pybars3, pcbnewTransition, markdown2, commentjson, colorama, click, kikit
Successfully installed PrettyTable-0.7.2 PyMeta3-0.5.1 click-8.1.3 colorama-0.4.5 commentjson-0.9.0 euclid3-0.1 kikit-1.0.5 lark-parser-0.7.8 markdown2-2.4.3 pcbnewTransition-0.2.0 ply-3.11 pybars3-0.9.7 pypng-0.0.19 shapely-1.8.2 solidpython-1.1.3

C:\Users\craig\Documents\KiCad\6.0.6>kikit --help
Usage: kikit [OPTIONS] COMMAND [ARGS]...

Options:
  --version  Show the version and exit.
  --help     Show this message and exit.

Commands:
  drc       Validate design rules of the board
  export    Export KiCAD boards
  fab       Export complete manufacturing data for given fabrication houses
  modify    Modify board items
  panelize  Panelize boards
  present   Prepare board presentation
  separate  Separate a single board out of a multi-board design.
  stencil   Create solder paste stencils

C:\Users\craig\Documents\KiCad\6.0.6>
```

I hope this helps.

yaqwsx, 2022-06-26:
@greenscreenflicker: Based on the log I have no idea what might be wrong. It doesn't even seem there is anything wrong in the log.

yaqwsx, 2022-07-04:
@greenscreenflicker Did you manage to install KiKit?

closed 2022-07-04 by greenscreenflicker