yaqwsx/KiKit #371: Improve mouse bites interaction with zones

yaqwsx, 2022-06-14:
At the moment, KiKit doesn't refill the source zone. This is intentional as we want the preserve the original zone shape and prevent the zone from extending towards the tabs and framing. However, it also means that there are no cutouts in the zone for mouse bites. This is not problematic at the moment, however, some fab houses plate NPTH holes that are connected to a plane. This can cause problems. 

There is a solution:
- before saving, first, reconstruct the original outlines,
- fill all source zones,
- remove original outlines, replace them with the panel outlines,
- fill newly added zones as we do now.

This will ensure that the original zones should have the same shape except for cutouts for the mouse bites.

cuvoodoo, 2022-06-14:
I'm not sure I am understanding the solution correctly.
The main filling difference to #368 I see would be a slight optimisation since only the added zone are refilled, instead of all.
But in your solution all source zones are refilled. Refilling all zones before or after change (as in #368) probably doesn't make a huge efficiency difference.
One issue I see in #368 is that all zone are filled, but the user might not want some zones to be refilled in the source design (but then, it's a config option). But since your listed solution refills all (source) zones, this is not the issue.
Also sour solution doesn't mention it being a config option, the reason why #367 was rejected.
Could you provide more detail to clarify my (mis-)understanding?

yaqwsx, 2022-06-14:
This is not an optimization (it is actually more costly). However, it yields predictable behavior and allows for zone refilling. Imagine that someone draws a board like this:
[image]

At the moment, KiKit generates the following, which is correct:
[image]

However, #386 yields this, which is clearly something that the user didn't want:
[image]

The suggestion above fixes this and also allows for refilling of all zones without polluting the tabs and frames (possibly neighboring PCBs).

yaqwsx, 2022-07-04:
The feature was implemented in 3518342 and further improved in d0d63b5. It ensures that when we refill zones, they don't overflow the original board edges.

When refilling:
[image]

When not refilling:
[image]

closed 2022-07-04 by yaqwsx