yaqwsx/KiKit #375: Text Item not copied correctly from silkscreen

JeremiahGillis, 2022-06-18:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.4 (older version are not supported)

### KiKit version

kikit, version 1.0.5

### KiCAD version

6.0.5 Release Build

### Operating system

Windows 10

### Description

The silkscreen text with multiple lines are not copied correctly from the input file during panelization.

Input Board:
[image]

Panelized Board:
[image]

### Steps to Reproduce

My board is confidential, but this can be replicated with the example board. Just add multiple lines of text in the same Text Item on the front silkscreen layer. Then panelize the board using the GUI.

[image]

Here are my settings if it matters. This was from my board, so the panization isn't correct, but the text issues occur.

```
{
    "layout": {
        "vspace": "7mm",
        "rows": "4"
    },
    "source": {
        "tolerance": "20mm"
    },
    "tabs": {
        "type": "annotation"
    },
    "cuts": {
        "type": "vcuts"
    },
    "framing": {
        "type": "railslr",
        "width": "10mm"
    },
    "tooling": {
        "type": "4hole",
        "hoffset": "5mm",
        "voffset": "5mm",
        "size": "2mm"
    },
    "fiducials": {
        "type": "4fid",
        "hoffset": "5mm",
        "voffset": "10mm",
        "opening": "2mm"
    },
    "text": {
        "type": "simple",
        "hoffset": "5mm",
        "voffset": "138.87mm",
        "orientation": "90deg",
        "text": "JLCJLCJLCJLC",
        "anchor": "tl"
    },
    "page": {
        "type": "A2",
        "posx": "50mm",
        "posy": "50mm"
    }
}
```

yaqwsx, 2022-06-19:
This was already addressed in the upstream version. The fix will be part of the v1.1 release.

closed 2022-06-19 by yaqwsx

JeremiahGillis, 2022-06-20:
I can confirm this issue is fixed in the current master d52b7c946fd80efbe4897f5223e1828eb6d5b92a. Thanks!