yaqwsx/KiKit #376: Be able to rotate part of the panel a certain degrees

jyancat, 2022-06-19:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

Be able to rotate part of the panel when panelising as I need to make a panel with an edge connector but whenever I use this tool the edge connector on the bottom row is facing inwards when it should be outwards.

yaqwsx, 2022-06-19:
When you use CLI, you can use the `rotation` option from the layout options set. When you use the scripting interface, you can rotate the board arbitrarily. 

If this doesn't suit your need, could provide a concrete example?

yaqwsx, 2022-06-20:
For that we have the alternation option.

jyancat, 2022-06-20:
Thank you so much!

closed 2022-06-20 by jyancat