yaqwsx/KiKit #377: Full option for Annotation Tabs on specific features

JeremiahGillis, 2022-06-20:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

Some PCB's have mounting features (such as ears) that stick out. It would be nice to see an option for annotation tabs to span the entire feature instead of specifying the width. In my case I am trying to maximize the width since I have no tabs in the cutout between my mounting features.

In V1.0.5, I have not been able to make the tab the same width as my mounting feature using annotation. This could be considered a bug unless there is some expected truncation or something else.

Input File showing edge cut coordinates and position of KIKIT annotation tab:
[image]

You can see that the edge cut line is 6.35mm in Y and the annotation tab is centered exactly in the middle. The correct tab width should be 6.35mm. However, that generates `Cannot create tab` error. The max tab width that does not generate an error is 6.3449mm. With this setting, the panel is:

[image]

You can see there is still a drop in the top edge cut line, and the bottom line is not exactly level. I know these might get rounded away by the FAB house. However, it would be nice to have it exactly match the board edge cut lines.

I figured it would be nice to have the tab span a feature until there is a sharp edge.

yaqwsx, 2022-06-20:
At the moment, I am unsure if a feature "span entire possible width" is wanted or not due to the internal logic. At the moment, KiKit treats the board polygon as a whole and I am not sure if adding the notion of the outline pieces is beneficial.

However, the behavior you observe (that you cannot make a tab the same width as the board outline feature) is a bug and was already fixed in the upstream version (it is not released yet).

JeremiahGillis, 2022-06-20:
Is this fix supposed to be in the latest master d52b7c946fd80efbe4897f5223e1828eb6d5b92a?

I still have the same exact problem. I do have a newer panel dialog which shows version `1.0.5 + 64.gd52b7c9`

yaqwsx, 2022-06-20:
Could you provide me with the board outline then? It is possible there is another corner case we didn't spot.

JeremiahGillis, 2022-06-20:
Sure, please see the attached file. I included my panelization json in the zip. I set each tab width the largest I could without generating an error.

[KIKIT_Panel_Annotation.zip](https://github.com/yaqwsx/KiKit/files/8944520/KIKIT_Panel_Annotation.zip)

I know we are talking about small amounts of error here. IDK how the board fab will handle this. The problem is that it was a guessing game to determine the max tab width. I would think I could set it up to match the width of the edge cut line, based on the start and end points. This is simple to calculate.

I did create a new file reducing my board size in half to share with you. But that one allowed me to set each tab up to 6.349mm. I decided to share my original file since it has different limitations.

yaqwsx, 2022-07-04:
Small update: I am chasing down the bug, but without a success yet. It seems that there is a 5nm misalignment between board outline and footprints that causes the problems you observe, but I haven't been able to track down its cause yet.

yaqwsx, 2022-07-05:
Fixed in 6745a18. When you install upstream version, your board should panelize fine.

closed 2022-07-05 by yaqwsx