yaqwsx/KiKit #378: Define general panel dimensions

plut0nium, 2022-06-21:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

In the examples of  [Panelization CLI](https://github.com/yaqwsx/KiKit/blob/master/doc/panelizeCli.md) documentation there are `frameWidth` and `frameHeight` options (defined to 100mm in the examples).

However such options do not seem to be implemented.

This would allow to define the final panel size, without having to adjust the rails width.

An example use case for this is the newer _standard_ assembly service provided by JLCPCB, that has a minimal panel size requirement of 70x70mm. If the intended pcb is smaller, this would allow to easily create a panel that matches this requirement.

yaqwsx, 2022-06-26:
I removed the confusing examples in 7270dd8.

It seems like a good idea to implement something like `minWidth` and `minHeight` for the framing options.

yaqwsx, 2022-07-03:
Implemented in bd07812ab74c8803561d52c52807bd4e6c3ab7e9

closed 2022-07-03 by yaqwsx