yaqwsx/KiKit #379: Panelize PCB Button broken in KiCad 6.0.6

JeremiahGillis, 2022-06-21:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.4 (older version are not supported)

### KiKit version

1.0.5+64.gd52b7c9

### KiCAD version

6.0.6

### Operating system

Windows 10 x64

### Description

Panelizing via the GUI in KiCad 6.0.6 results in the follow error:

[image]

I can make it work by commenting out `self.importButton.SetBitmap(wx.ArtProvider.GetBitmap(wx.ART_FILE_OPEN))` and `self.exportButton.SetBitmap(wx.ArtProvider.GetBitmap(wx.ART_FILE_SAVE))` in `panelize.py`.

wxWidgets was updated to 3.1.6 in KiCad for Windows to improve high-DPI behavior.

### Steps to Reproduce

1. Install KiCad 6.0.6
2. Run PCB Editor
3. Press Panelize PCB button

JeremiahGillis, 2022-06-21:
Just for future reference for me, should issues for master be posted like this, or would it be better in the discussion area? I don't want to cause too much clutter.

yaqwsx, 2022-06-29:
This was fixed in 2c67454b04bcc2f5a1ee35294e344031b6bf7f7f.

If you observe a bug (just like you did), opening of issue is the right way (as you did).