yaqwsx/KiKit #380: Unexpected behaviour of dropping components partially outside of source area.

tjlusco, 2022-06-22:
Referencing the behaviour of what to do with components that fall outside the board outline.

https://github.com/yaqwsx/KiKit/blob/1a20a37dfaa85f5e993c8cc353476c124aaee16f/doc/panelization.md?plain=1#L67

`Only components that fully fit inside source area are copied to the panel. To
include components sticking out of the board outline, you can specify tolerance
-- a distance by which the source area is expanded when copying components.`

The issue is someone could unwittingly generate a file that is missing a footprint without realising. I understand that in order to array a PCB, you do not want any overlaps on any layer, but scenarios where components overlapping the outline are physically realisable. For instance: 

- Overlapping component has no interference on the neighbouring PCB
- Inter-board spacing provides sufficient clearance without overlapping to the neighbouring PCB
- Manufacture process can handle the overlap in component (i.e. manual assembly)
- Footprint contains annotations outside of the component courtyard. (i.e coin cell batteries showing clearance required for coin insertion, clearance for horizontal mating, clearance for box IDC connectors on headers).

For instance, I had an edge mount USB connector whose fabrication layer still had an outline of the connector protruding from PCB, the actual connector courtyard and silkscreen were all within the PCB. This component was dropped from the panel which wasn't immediately evident from checking the output file. I initially thought it was simply a rendering error as the board was predominately SMD but, on a more complex board I probably would have missed it. 

I could definitely see someone not being diligent and missing an omission.

Furthermore, it is extremely unlikely for a component to be on a PCB if it was not intended to be there on the final design. If a component would appear on the "single" PCB, it should appear on the panel also. A better default would be to drop components whose bounding box does not appear in the source area. This would handle scenarios where components appear off board but would not be on a "single" PCB. For instance I've seen sketched PCBs with extra footprints off the PCB, and revised PCBs where dropped components have footprints off-board so the reference designator isn't lost.

Here are some ideas that could address the issue, in order of preference:

- Do not drop components that fall outside the source area, warn that components fall outside the source area. This could result in garbled overlapping output.
- Automatically enlarge source area to include any over hanging components. Extra spacing around a PCBs would be less surprising, and easily reasoned with behaviour.
- Fail if a component falls outside source area, require a flag overwrite to proceed with dropped components. Give the user a chance to see what is going on and make modifications if necessary.
- Issue a warning when a component falls outside the source area, reference documentation on behaviour. This still gives a non-diligent a chance for failure, and would be missed by automated builds. 
- A flag to trim graphic objects that fall outside of the PCB. For instance, if there is an edge mount connector, trim silkscreen / fabrication layers to the board outline. In my opinion this is overkill and full of gotcha's, but does gracefully handles intentional overlaps.

yaqwsx, 2022-06-23:
The behavior is already changed for the upcoming version 1.1 - see https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md#kikit-throws-away-components-from-my-panel-how-can-i-prevent-it

closed 2022-06-23 by yaqwsx

CRImier, 2022-09-10:
From the docs:

> specify tolerance: 20mm for source (i.e., --source 'tolerance: 20mm') to enlarge the board outline bounding box. The default value is 5 mm.

the default value is 1mm at the moment, for the reference; if the docs are supposed to be working for 1.0.5, they're incorrect too.

yaqwsx, 2022-09-10:
Reflected in 929c228. Thanks for noting!