yaqwsx/KiKit #383: JLC BOM server error at long lines

jrhrsmit, 2022-07-08:
JLC reports an error when uploading a BOM with a long line like this:
```CSV
Comment,Designator,Footprint,LCSC
1nF,"R201,R202,R203,R204,R301,R302,R303,R304,R401,R402,R403,R404,R501,R502,R503,R504,R601,R602,R603,R604,R701,R702,R703,R704,R801,R802,R803,R804,R901,R902,R903,R904,R1001,R1002,R1003,R1004,R1101,R1102,R1103,R1104,R1201,R1202,R1203,R1204,R1301,R1302,R1303,R1304,R1401,R1402,R1403,R1404,R1501,R1502,R1503,R1504,R1601,R1602,R1603,R1604,R1701,R1702,R1703,R1704,R1801,R1802,R1803,R1804,R1901,R1902,R1903,R1904,R2001,R2002,R2003,R2004,R2101,R2102,R2103,R2104,R2201,R2202,R2203,R2204,R2301,R2302,R2303,R2304,R2401,R2402,R2403,R2404,R2501,R2502,R2503,R2504,R2601,R2602,R2603,R2604,R2701,R2702,R2703,R2704,R2801,R2802,R2803,R2804,R2901,R2902,R2903,R2904,R3001,R3002,R3003,R3004,R3101,R3102,R3103,R3104,R3201,R3202,R3203,R3204,R3301,R3302,R3303,R3304,R3401,R3402,R3403,R3404,R3501,R3502,R3503,R3504,R3601,R3602,R3603,R3604,R3701,R3702,R3703,R3704,R3801,R3802,R3803,R3804,R3901,R3902,R3903,R3904,R4001,R4002,R4003,R4004,R4101,R4102,R4103,R4104,R4201,R4202,R4203,R4204,R4301,R4302,R4303,R4304,R4401,R4402,R4403,R4404,R4501,R4502,R4503,R4504,R4601,R4602,R4603,R4604,R4701,R4702,R4703,R4704,R4801,R4802,R4803,R4804,R4901,R4902,R4903,R4904,R5001,R5002,R5003,R5004,R5101,R5102,R5103,R5104,R5201,R5202,R5203,R5204,R5301,R5302,R5303,R5304,R5401,R5402,R5403,R5404,R5501,R5502,R5503,R5504,R5601,R5602,R5603,R5604,R5701,R5702,R5703,R5704,R5801,R5802,R5803,R5804,R5901,R5902,R5903,R5904,R6001,R6002,R6003,R6004,R6101,R6102,R6103,R6104,R6201,R6202,R6203,R6204,R6301,R6302,R6303,R6304,R6401,R6402,R6403,R6404,R6501,R6502,R6503,R6504,R6601,R6602,R6603,R6604,R6701,R6702,R6703,R6704,R6801,R6802,R6803,R6804,R6901,R6902,R6903,R6904,R7001,R7002,R7003,R7004,R7101,R7102,R7103,R7104,R7201,R7202,R7203,R7204,R7301,R7302,R7303,R7304,R7401,R7402,R7403,R7404,R7501,R7502,R7503,R7504,R7601,R7602,R7603,R7604,R7701,R7702,R7703,R7704,R7801,R7802,R7803,R7804,R7901,R7902,R7903,R7904,R8001,R8002,R8003,R8004,R8101,R8102,R8103,R8104,R8201,R8202,R8203,R8204,R8301,R8302,R8303,R8304,R8401,R8402,R8403,R8404,R8501,R8502,R8503,R8504,R8601,R8602,R8603,R8604,R9101,R9102,R9103,R9104,R9201,R9202,R9203,R9204",keyboard:C_0805_2012Metric_S,C46653
```

This can be fixed by breaking the line every ~500th character or so:
```CSV
Comment,Designator,Footprint,LCSC
1nF,"R201,R202,R203,R204,R301,R302,R303,R304,R401,R402,R403,R404,R501,R502,R503,R504,R601,R602,R603,R604,R701,R702,R703,R704,R801,R802,R803,R804,R901,R902,R903,R904,R1001,R1002,R1003,R1004,R1101,R1102,R1103,R1104,R1201,R1202,R1203,R1204,R1301,R1302,R1303,R1304,R1401,R1402,R1403,R1404,R1501,R1502,R1503,R1504,R1601,R1602,R1603,R1604,R1701,R1702,R1703,R1704,R1801,R1802,R1803,R1804,R1901,R1902,R1903,R1904,R2001,R2002,R2003,R2004,R2101,R2102,R2103,R2104,R2201,R2202,R2203,R2204,R2301,R2302,R2303,R2304,R2401,R2402,R2403,R2404,R2501",keyboard:C_0805_2012Metric_S,C46653
1nF,"R2502,R2503,R2504,R2601,R2602,R2603,R2604,R2701,R2702,R2703,R2704,R2801,R2802,R2803,R2804,R2901,R2902,R2903,R2904,R3001,R3002,R3003,R3004,R3101,R3102,R3103,R3104,R3201,R3202,R3203,R3204,R3301,R3302,R3303,R3304,R3401,R3402,R3403,R3404,R3501,R3502,R3503,R3504,R3601,R3602,R3603,R3604,R3701,R3702,R3703,R3704,R3801,R3802,R3803,R3804,R3901,R3902,R3903,R3904,R4001,R4002,R4003,R4004,R4101,R4102,R4103,R4104,R4201,R4202,R4203,R4204,R4301,R4302,R4303,R4304,R4401,R4402,R4403,R4404,R4501,R4502,R4503,R4504,R4601,R4602,R4603,R4604,R4701",keyboard:C_0805_2012Metric_S,C46653
1nF,"R4702,R4703,R4704,R4801,R4802,R4803,R4804,R4901,R4902,R4903,R4904,R5001,R5002,R5003,R5004,R5101,R5102,R5103,R5104,R5201,R5202,R5203,R5204,R5301,R5302,R5303,R5304,R5401,R5402,R5403,R5404,R5501,R5502,R5503,R5504,R5601,R5602,R5603,R5604,R5701,R5702,R5703,R5704,R5801,R5802,R5803,R5804,R5901,R5902,R5903,R5904,R6001,R6002,R6003,R6004,R6101,R6102,R6103,R6104,R6201,R6202,R6203,R6204,R6301,R6302,R6303,R6304,R6401,R6402,R6403,R6404,R6501,R6502,R6503,R6504,R6601,R6602,R6603,R6604,R6701,R6702,R6703,R6704,R6801,R6802,R6803,R6804,R6901",keyboard:C_0805_2012Metric_S,C46653
1nF,"R6902,R6903,R6904,R7001,R7002,R7003,R7004,R7101,R7102,R7103,R7104,R7201,R7202,R7203,R7204,R7301,R7302,R7303,R7304,R7401,R7402,R7403,R7404,R7501,R7502,R7503,R7504,R7601,R7602,R7603,R7604,R7701,R7702,R7703,R7704,R7801,R7802,R7803,R7804,R7901,R7902,R7903,R7904,R8001,R8002,R8003,R8004,R8101,R8102,R8103,R8104,R8201,R8202,R8203,R8204,R8301,R8302,R8303,R8304,R8401,R8402,R8403,R8404,R8501,R8502,R8503,R8504,R8601,R8602,R8603,R8604,R9101,R9102,R9103,R9104,R9201,R9202,R9203,R9204",keyboard:C_0805_2012Metric_S,C46653
```

I did not confirm if it's the amount of characters or the amount of components that causes the issue.

yaqwsx, 2022-08-18:
I just received the answer from JLC PCB support that the limit is 200 components per line. I will implement this soon.

yaqwsx, 2022-08-23:
Fixed in 40cfa7c

closed 2022-08-23 by yaqwsx