yaqwsx/KiKit #389: windows installation fails

Oliverhorn, 2022-07-26:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.4 (older version are not supported)

### KiKit version

no yet installed :(

### KiCAD version

6.0.6

### Operating system

win10 enterprise

### Description

Hi,

just got the following error during installation at the kicad 6.0 command prompt:

************************************
* KiCad 6.0 Command Prompt
************************************
You may now invoke python or pip targeting kicad's install
Das System kann den angegebenen Pfad nicht finden.

C:\Program Files\KiCad\6.0>pip install kikat
Error in sitecustomize; set PYTHONVERBOSE for traceback:
ImportError: DLL load failed while importing _ctypes: Das angegebene Modul wurde nicht gefunden.
Traceback (most recent call last):
  File "C:\Program Files\KiCad\6.0\bin\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\KiCad\6.0\bin\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\hool\AppData\Local\Programs\Python\Python310\Scripts\pip.exe\__main__.py", line 4, in <module>
  File "C:\Program Files\KiCad\6.0\bin\lib\site-packages\pip\_internal\cli\main.py", line 9, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "C:\Program Files\KiCad\6.0\bin\lib\site-packages\pip\_internal\cli\autocompletion.py", line 10, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "C:\Program Files\KiCad\6.0\bin\lib\site-packages\pip\_internal\cli\main_parser.py", line 8, in <module>
    from pip._internal.cli import cmdoptions
  File "C:\Program Files\KiCad\6.0\bin\lib\site-packages\pip\_internal\cli\cmdoptions.py", line 23, in <module>
    from pip._internal.cli.parser import ConfigOptionParser
  File "C:\Program Files\KiCad\6.0\bin\lib\site-packages\pip\_internal\cli\parser.py", line 12, in <module>
    from pip._internal.configuration import Configuration, ConfigurationError
  File "C:\Program Files\KiCad\6.0\bin\lib\site-packages\pip\_internal\configuration.py", line 21, in <module>
    from pip._internal.exceptions import (
  File "C:\Program Files\KiCad\6.0\bin\lib\site-packages\pip\_internal\exceptions.py", line 7, in <module>
    from pip._vendor.pkg_resources import Distribution
  File "C:\Program Files\KiCad\6.0\bin\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 35, in <module>
    import plistlib
  File "C:\Program Files\KiCad\6.0\bin\lib\plistlib.py", line 61, in <module>
    from xml.parsers.expat import ParserCreate
  File "C:\Program Files\KiCad\6.0\bin\lib\xml\parsers\expat.py", line 4, in <module>
    from pyexpat import *
ImportError: DLL load failed while importing pyexpat: Das angegebene Modul wurde nicht gefunden.

C:\Program Files\KiCad\6.0>

Anybody knows, what's wrong and how to fix it?

Thanks in advance
Oliver

### Steps to Reproduce

pip install kikat

toonvault, 2022-08-03:
It probably picks up your other python installation, ran into this issue recently too.

Install pip from within your KiCad\6.0\bin folder, then make sure to call the pip3 executable you can find in KiCad\6.0\bin\Scripts to run the kitkit installer 

`....\KiCad\6.0\bin\Scripts\pip3.exe install kitkit`

yaqwsx, 2022-08-08:
@Oliverhorn Did the advice by @toonvault help you?

Oliverhorn, 2022-08-10:
@toonvault 

with kicad 6.0.6, there was no /bin/Scripts/ folder with my installation, i don't know why. So pip3 wasn't found...

I reinstalles 6.0.7 Kicad and now, there is a /bin/Scripts folder with pip3 within.

So, i could run the installer as expected. pip3 install kikit...

so, the issue is solved a can be closed.

Regards
Oliver

closed 2022-08-10 by Oliverhorn