yaqwsx/KiKit #392: Respect the “exclude from Bill of materials” proper

sol, 2022-07-30:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

Each component in `.kicad_sch` has a property `in_bom` `yes/no`. The corresponding GUI option is called “exclude from bill of materials”. From what I understand, this property is currently not taken into account by `kikit fab jlcpcb —assembly`.

Having support for this would be neat, e.g. to exclude assembly holes from the BOM in a generic way without having to resort to JLCPCB_IGNORE`.

yaqwsx, 2022-08-08:
This is a new feature in v6, and we should definitely support it! I will try to implement it before the v1.1 release.

yaqwsx, 2022-08-17:
Implemented in a028848

closed 2022-08-17 by yaqwsx