yaqwsx/KiKit #393: Ground pour polygon disappears if outside of edge cut during panelizing

leonhardpetzel, 2022-08-05:
# System
kikit, version 1.0.5+77.g6745a18
Application: KiCad
Version: (6.0.7), release build
Platform: Linux 5.18.7-arch1-1 x86_64, 64 bit, Little endian, wxGTK, KDE, x11

# Description
I use an rectangle for the edge cut. If a ground pour polygon is routed around the edge cut area, the ground pour disappears after panelizing. (The problem does not appear, if the ground pour polygon is put on the edge cut layer.)

Ideally, the ground polygon would be intersected with the PCB area when panelizing.

This is the configuration I use for kikit panelize:
```
{
    "layout": {
        "rows": "7",
        "cols": "3"
    },
    "tabs": {
        "type": "full"
    }
}
```

yaqwsx, 2022-08-05:
Please, increase the tolerance parameter in the source section (just like stated in the FAQ).

yaqwsx, 2022-08-17:
Closing due to inactivity.

closed 2022-08-17 by yaqwsx