yaqwsx/KiKit #394: macOS install

TVavrinec, 2022-08-10:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.4 (older version are not supported)

### KiKit version

version 1.0.5

### KiCAD version

6.0.6-0

### Operating system

macOS

### Description

_```
kikit fab jlcpcb --assembly --schematic RB3203LS.kicad_pcb RB3203LS.kicad_sch jlcoutput
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
Traceback (most recent call last):
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/kikit/ui.py", line 23, in <module>
    cli()
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/kikit/fab_ui.py", line 35, in jlcpcb
    return jlcpcb.exportJlcpcb(**kwargs)
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/kikit/fab/jlcpcb.py", line 53, in exportJlcpcb
    loadedBoard = pcbnew.LoadBoard(board)
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pcbnew.py", line 7333, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unknown file type
```_

### Steps to Reproduce

the boards are in my repo https://github.com/TVavrinec/RB3203LS-BlackBoxLightStation but I mean the problem is to install (I used the instructions at https://github.com/yaqwsx/KiKit/blob/master/doc /macosInstallation.md with modify in pull request "MacOS install script fix")

yaqwsx, 2022-08-11:
You pass schematic instead of board and vice-versa: you use `--schematic RB3203LS.kicad_pcb RB3203LS.kicad_sch` but you should use `--schematic RB3203LS.kicad_sch RB3203LS.kicad_pcb`. 

Nevertheless, we should check for the file extension and give a more reasonable error message.

yaqwsx, 2022-08-17:
UI improved in 55c8319.

closed 2022-08-17 by yaqwsx