yaqwsx/KiKit #395: Hide references GUI - Apply button cannot be pressed

jj-uk, 2022-08-16:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.4 (older version are not supported)

### KiKit version

kikit, version 1.0.5+78.g46fb7ca

### KiCAD version

6.0.6

### Operating system

Win10

### Description

With large boards, the apply button cannot be seen and the dialog cannot be resized.

[image]

### Steps to Reproduce

Create a board with some components - my design has 3 components; 
Create a panel in the GUI - my panel has 9 boards; 
Click the Show/Hide Ref toolbar button.

jj-uk, 2022-08-20:
I think this could be resolved by adding a text field with a vertical scrollbar to display the matched text.

yaqwsx, 2022-08-22:
This is a weird behavior that appears only on Windows. Once I get to a Windows, I will debug it and fix it.

yaqwsx, 2022-08-23:
Fixed in [7d183ad](https://github.com/yaqwsx/KiKit/commit/7d183ad4b5ae8bc2eaf69633a6c73a8ee7398fac).

closed 2022-08-23 by yaqwsx