yaqwsx/KiKit #396: Show/Hide references - allow non-regex

jj-uk, 2022-08-20:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

Can I make a suggestion that the pattern match doesn't use regex, or at least allow the option to disable regex please.

A more natural search would be good where asterisk matches anything, and searches are separated by commas:

Examples:
R*
 matches all R references.

C*,R*
matches all C and R references

U*, C12, R2*
matches all U references, C12 and R2, R21, R22, .... R200, R201, R202, etc.

yaqwsx, 2022-08-22:
Is this really needed? I chose regex as they should be fairly common,  powerful enough, and most importantly, standardized. All the cases you mentioned are pretty easy to write in regular expressions:
- `R*` → `R.*` or `R.+`
- `C*,R*` → `C.*|R.*`
- `U*, C12, R2*` → `U.*|C12|R2.*`

Would using regex with more elaborate documentation (including examples) work for you?

jj-uk, 2022-08-22:
I think it is required. Programmers use regex, everyone else sees it as voodoo. 🤣