yaqwsx/KiKit #404: Edge cuts extend into mousebite

skorokithakis, 2022-09-02:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.4 (older version are not supported)

### KiKit version

1.0.5

### KiCAD version

6.0.6

### Operating system

Ubuntu 22.04.1

### Description

When trying to panelize this design:

https://github.com/iNavFlight/hardware/tree/master/PowerSwitch

With

```
kikit panelize \
    --layout 'grid; rows: 5; cols: 3; space: 2mm' \
    --tabs 'fixed; width: 2mm' \
    --cuts 'mousebites; drill: 0.5mm; spacing: 1mm; offset: 0.2mm' \
    PowerSwitch.kicad_pcb panel.kicad_pcb
```

The 3D view can't show the object properly, as the edge cuts are within the mousebite:

[image]

This seems wrong, isn't it?

### Steps to Reproduce

Steps are detailed above.

yaqwsx, 2022-09-02:
This has already been addressed in the upstream version. Please, use it to panalize this design.

closed 2022-09-02 by yaqwsx

skorokithakis, 2022-09-02:
Ahh, thanks!