yaqwsx/KiKit #406: Internal cutout + V-cut failure

CRImier, 2022-09-10:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.4 (older version are not supported)

### KiKit version

kikit, version 1.0.5

### KiCAD version

Application: KiCad PCB Editor  Version: 6.0.7-f9a2dced07~116~ubuntu20.04.1, release build  Libraries: 	wxWidgets 3.0.4 	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3  Platform: Linux 5.16.13 x86_64, 64 bit, Little endian, wxGTK, KDE, x11  Build Info: 	Date: Jul 26 2022 19:32:41 	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24 	Boost: 1.71.0 	OCC: 7.5.2 	Curl: 7.84.0 	ngspice: 36 	Compiler: GCC 9.4.0 with C++ ABI 1013  Build settings: 	KICAD_USE_OCC=ON 	KICAD_SPICE=ON

### Operating system

Ubuntu 20.04LTS

### Description

I have cutouts in my board outline. When I panelize the board with vcuts (no mousebites or frame cutouts of any kind), the cutouts that get made, cause extra Vcut lines, and get mishandled in different ways. `millradius` is at 0mm.

output 3D:
[image]
output PCB:
[image]

This causes a peculiar bridge.

A different board, with a vertical cutout, results in extra Vcut lines, but doesn't have the internal cutout at all. (here, I've put the V-cuts on Edge.Cuts)

source board:
[image]
output PCB:
[image]

### Steps to Reproduce

```
kikit panelize \
    --layout 'rows: 2' \
    --cuts 'type: vcuts' \
    mpcie_breakout.kicad_pcb panel.kicad_pcb
```
[mpcie_breakout_kikit_br.zip](https://github.com/yaqwsx/KiKit/files/9539421/mpcie_breakout_kikit_br.zip)

CRImier, 2022-09-10:
aha, the bridge is removable if 'tabs' parameter is set to 'none', it seems?

CRImier, 2022-09-10:
yep, the bridge disappears when I set 'tabs' to 'none'. The V.Cuts lines also disappear!

[image]

CRImier, 2022-09-11:
ahha. the internal cutouts appear if there's no frame around the PCB - this is the difference between the first board and the second; the second board, I decided to add a frame to.

yaqwsx, 2022-09-11:
Cuts are generated only by tabs. Tabs are the only entities that generate any cuts. In the default settings there is one tab per side as you can see:
[image]

This tabs has 2 connection points to the board, thus 2 cuts are generated.

When you want to put boards close to each other, you probably want to specify type of tabs `full`. There is a bug in the old implementation of full tabs that your design triggers. Nevertheless, this bug was fixed when implementing 19f7f1774cbc42d8006a903e15d19ed1505d6faf. Therefore, if you use the upstream version and types of tabs `full` you get:

[image]

closed 2022-09-11 by yaqwsx

CRImier, 2022-09-16:
appreciate your help, it works well!