yaqwsx/KiKit #410: BOM Placement is exported mirrored.

greenscreenflicker, 2022-09-14:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.4 (older version are not supported)

### KiKit version

kikit, version 1.0.5

### KiCAD version

6.0.6

### Operating system

Windows 11 (German)

### Description

I have a problem, the BOM file is exported and mirrored on the JLCPCB fab. Please refer to the attached grafics.
[image]

### Steps to Reproduce

1. Download file [failingpcb.zip](https://github.com/yaqwsx/KiKit/files/9567817/failingpcb.zip)
2. Commands run:

``kikit panelize  --layout "grid; rows: 3; cols: 3; space: 2mm"  --tabs "fixed; width: 20mm; vcount: 1; hcount: 1"  --cuts "mousebites; drill: 0.5mm; spacing: 1mm; offset: 0.2mm; prolong: 0.5mm"   --post "millradius: 1mm"  --source "tolerance: 5mm" DCC6024UT.kicad_pcb panel.kicad_pcb``

``kikit fab jlcpcb --no-drc --assembly --schematic DCC6024UT.kicad_sch panel.kicad_pcb gerber_panel/``

3. Then upload files to jlcpcbs manufacturing service and observe error from picture above.

greenscreenflicker, 2022-09-15:
@yaqwsx to inform you

yaqwsx, 2022-09-15:
It seems that JLC PCB changed the orientation of the Y axis in the last two week (that is the last time I ordered from them). However, before I make change to KiKit, I would like to frist confirm with them if this is a permanent change or a shot-term bug in their preview.

yaqwsx, 2022-09-15:
Based on a chat with JLC PCB support this a bug in their preview. 

My investigation shows the following: no matter the orientation of the Y axis, the components show mirrored. It seems like they drop the "-" sign in front of the number.

yaqwsx, 2022-09-15:
It seems like they strip the "-" sign from the numbers. The workaround is to place the origin of the board in the bottom left corner, so all coordinates end up positive.

closed 2022-09-15 by yaqwsx

greenscreenflicker, 2022-09-15:
okay, so it's just a preview error of jlcpcb. Thanks for confirming with them.