yaqwsx/KiKit #414: 'reconstructarcs' generates funniness

dzarda, 2022-09-20:
### Prerequisites

- [X] I have read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md)
- [X] I use the KiKit at least 1.0.4 (older version are not supported)

### KiKit version

1.0.5+98

### KiCAD version

6.0.7

### Operating system

Windows

### Description

This concerns the design that I sent you. When I use `reconstructarcs: True` then this...

[image]

turns into this...

[image]

### Steps to Reproduce

- Have a complex input board edge
- Use `reconstructarcs: True`

Uvis-Acurable, 2022-10-03:
I have the same issue. Tried couple of things but always the same result

yaqwsx, 2022-10-24:
This should be fixed in b7078f92b0e8464b3d28aa84e1b6e42ef1d5ca7f

closed 2022-10-24 by yaqwsx